agronick/aa-torque #15: [BUG]

apristel, 2023-10-04:
Pixel 7 Pro android 14 - follow3d instructions several times, does not show in AA.

whitecrixu, 2023-10-04:
Is not a BUG app is not work on Pixel 7 Pro Android 13/14 I try tomorrow with root acces ;)

apristel, 2023-10-04:
Ok thank you! Wasn't sure what to classify it as.

jpcaldwell30, 2023-10-04:
I also am having the same issue. Can you clarify/update the instructions for the Pixel 7 pro once you have it updated/fixed? Really appreciate your work!

aross6, 2023-10-04:
Google Pixel 6a, Android 13, USA. Also followed the instructions exactly multiple times and AA Torque is not showing for me in Android Auto

ngknick, 2023-10-04:
Pixel 7 pro on Google Fi service, as vanilla as it gets. Have torque pro. Enabled auto developer mode. Enabled unknown sources. Granted permissions and chose settings for AAtorque apk, rebooted both devices still doesn't show on Android Auto.

SAABoholic, 2023-10-06:
While I have the same issue with my Pixels it worked just fine on a bone stock, S23 Ultra.

Keeno80, 2023-10-06:
Same issue on my pixel 7 running Android 13. Will try a different phone

ngknick, 2023-10-06:
For this phone you need to reinstall\update the current Google PackageInstaller, you can download the same version from ApkMirror or similar site, or you can get the apk directly from your phone using a file manager with that feature (I'm use mixplorer) .... Found this on the readme file for Kingstaller ... Can anyone help with the package installer update? No idea how to do that

ngknick, 2023-10-06:
> > For this phone you need to reinstall\update the current Google PackageInstaller, you can download the same version from ApkMirror or similar site, or you can get the apk directly from your phone using a file manager with that feature (I'm use mixplorer) .... Found this on the readme file for Kingstaller ... Can anyone help with the package installer update? No idea how to do that
> 
> No dice on Android 14 on Pixel 6a. Only 3 Package Installer apk's available for Android 14 and none of them allow KingInstaller to install the package "as Google Play store". And we're not able to install an older Package Installer version (from Android 13) because the SDK level required now is 34 (and the older Android 13 SDK levels are 33). I tried extensively last night with different Package Installer apk's as well as different versions of KingInstaller and AA Torque without success. All of them ended with AA Torque being installed as "by KingInstaller", which is why the app isn't showing up in Android Auto.

Same here, managed to find the package installers but AA torque still shows up as installed by King.. I noticed there's a message at the bottom of king that says to use no root method due to my package installer but I can't find any info on that

agronick, 2023-10-06:
This is a hack and as such the process on getting it running will be different on certain phones. It is not a problem with the code so is not a bug. Please move the discussion here: https://github.com/agronick/aa-torque/discussions/23

closed 2023-10-06 by agronick