agronick/aa-torque #17: [REQUEST] PSI to BAR conversion

FPSUsername, 2023-10-05:
**Is your feature request related to a problem? Please describe.**
In Jille's power monitor app, the boost is shown in bar. In aa torque, it shows up as PSI by default.

**Describe the solution you'd like**
Boost to default as BAR or a conversion added to the list that converts PSI to BAR, because it only exists as BAR to PSI.

**Describe alternatives you've considered**
Manually do the conversion, but it should be in the list.

**Additional context**
None

agronick, 2023-10-06:
If someone wants to edit the xml to add these, it is pretty easy. https://github.com/agronick/aa-torque/blob/master/app/src/main/res/values/user_scripts.xml

closed 2023-10-06 by agronick

spadab, 2023-10-09:
when I use this string  (psi to bar) the value is no longer shown.  How can it be fixed?  Thank you