agronick/aa-torque #18: [REQUEST] AA Torque name also in Android launcher

cioce78, 2023-10-05:
Hi, please rename the APP as "AA Torque" also in standard Android launcher, now is using old name "Setting Performance Monitor".

agronick, 2023-10-06:
You can help translate here: https://poeditor.com/projects/view?id=653187

Airislv24, 2023-10-19:
by this link https://poeditor.com/projects/view?id=653187
[image]

agronick, 2023-10-19:
Sorry this link https://poeditor.com/join/project/yttme0y1VZ

emiliodeme, 2023-12-08:
Can't help on translation as it returns project can't accept more contributors. Would like to help on a Portuguese (BR) translation.

agronick, 2023-12-13:
I contacted the people at the translation service. It should be fixed now. They re-enabled the features that were missing.

closed 2023-12-29 by agronick