agronick/aa-torque #21: Temperatures not reporting correctly

sinistriel, 2023-10-05:
When you select "whole numbers" it rounds the temperatures up by 20 plus degrees in the new releases, earlier releases did not have this issue. 

[image]
[image]

agronick, 2023-10-06:
What equation are you using for the conversion? Can you see it moving 20 degrees just by checking and unchecking the box?

sinistriel, 2023-10-06:
Yeah if I check or uncheck the box it will move 20~ degrees. Unchecked it
matches the display on Torque exactly then bumps up by 20~ when checked.

sinistriel, 2023-10-06:
I'm using the unit conversion, and this behavior started after the last few
updates, none of the betas had this issue.

agronick, 2023-10-06:
There was a bug, will be fixed in the next release

closed 2023-10-06 by agronick