agronick/aa-torque #33: ODBC adapter

DokVWTSi, 2023-10-09:
My ODBC adapter is Wi-Fi. Will it work or does the adapter need to be Bluetooth? I have the program opening but tells me "connecting to device.." going into Torque Pro the device works perfectly as soon as I have connected to the ELM Wi-Fi network. Love the interface and would really like to get it working. Thank you and well done with the dev. It's awesome

agronick, 2023-10-10:
It should work fine. Make sure your Torque is updated and you have configured your gauges.

Fabio1968, 2023-10-12:
If I understand correctly, can the OBD be connected via wifi without using the USB cable from the phone to the car?
thanks for your reply

agronick, 2023-10-12:
Yes, it is just asking Torque for data. If it works with Torque it should work here.

Fabio1968, 2023-10-12:
Thank you for your answer. Can you recommend a wifi OBD model? There are several models and I don't know exactly what features it should have. Thank you

agronick, 2023-10-12:
I use Bluetooth. I use the wifi for wireless android auto.

closed 2023-12-29 by agronick