agronick/aa-torque #39: [BUG] PSI to BAR custom expression

spadab, 2023-10-12:
After the release of version R5.1 the custom expression for convert psi to bar no longer works.  I have tried all subsequent versions but the turbo pressure data remains unchanged at zero.  
I used a / 14.5038
In relase R5.1 the custom conversion works perfectly.
Thanks for the wonderful work!

agronick, 2023-10-13:
Does it work when you turn unit conversion off? It works for me.

spadab, 2023-10-13:
Yes exactly, when I turn off the unit conversion works well and the instrument shows the pressure data only in psi.

agronick, 2023-10-14:
Can you post your logs? Select the menu on the top right and copy logs to clipboard.

spadab, 2023-10-15:
Where is the log file recorded?  The last working version was 5.1 what happened after it?

agronick, 2023-10-16:
It's in the menu on the settings page. A lot happened but not much with unit conversions.

spadab, 2023-10-16:
I can't find the log file on my phone, why don't I have root permissions?  however it app ears that the "name string" list at the bottom is missing from the 
[image]
app/src/main/res/values/user_scripts.xml file

spadab, 2023-10-16:
I can't find the log file on my phone, why don't I have root permissions?  
However the last part is missing from the user_script file?  
missing "name string" list at bottom?
app/src/main/res/values/user_scripts.xml 

[image]

agronick, 2023-10-16:
No they were just moved to the rest of the strings so they could be translated. There is an option for copying the logs when you tap the 3 dots on the settings page.

agronick, 2023-10-17:
If you set the custom expression to "a". Does it show anything? You need to paste the log.

elnidalarguia, 2023-10-17:
how do I copy the logs?

> Si establece la expresión personalizada en "a". ¿Muestra algo? Necesitas pegar el registro.

elnidalarguia, 2023-10-17:
TorqueRefresher$doRefresh: Debug Got valid 0.0 from torque for COM-BP-A 
TorqueRefresher$doRefresh: Debug Got valid -10.442717121373825 from torque for Boost 
TorqueGauge: Debug Settings max speed 0.0 
TorqueRefresher$doRefresh: Debug Got valid 0.0 from torque for HP 
TorqueRefresher$doRefresh: Debug Got valid 0.0 from torque for COM-BP-A 
TorqueRefresher$doRefresh: Debug Got valid -10.297679383576966 from torque for Boost 
TorqueGauge: Debug Settings max speed 0.0 
TorqueRefresher$doRefresh: Debug Got valid 0.0 from torque for HP 
TorqueRefresher$doRefresh: Debug Got valid 0.0 from torque for COM-BP-A 
TorqueRefresher$doRefresh: Debug Got valid -10.297679383576966 from torque for Boost 
TorqueGauge: Debug Settings max speed 0.0 
TorqueRefresher$doRefresh: Debug Got valid 0.0 from torque for HP 
TorqueRefresher$doRefresh: Debug Got valid 0.0 from torque for COM-BP-A 
TorqueRefresher$doRefresh: Debug Got valid -10.297679383576966 from torque for Boost 
TorqueGauge: Debug Settings max speed 0.0 
TorqueRefresher$doRefresh: Debug Got valid 0.0 from torque for HP 
TorqueRefresher$doRefresh: Debug Got valid 0.0 from torque for COM-BP-A 
TorqueRefresher$doRefresh: Debug Got valid -10.442717121373825 from torque for Boost 
TorqueGauge: Debug Settings max speed 0.0 
TorqueRefresher$doRefresh: Debug Got valid 0.0 from torque for HP 
TorqueRefresher$doRefresh: Debug Got valid 0.0 from torque for COM-BP-A 
TorqueRefresher$doRefresh: Debug Got valid -10.297679383576966 from torque for Boost 
TorqueGauge: Debug Settings max speed 0.0 
TorqueRefresher$doRefresh: Debug Got valid 0.0 from torque for HP 
TorqueRefresher$doRefresh: Debug Got valid 0.0 from torque for COM-BP-A 
TorqueRefresher$doRefresh: Debug Got valid -10.297679383576966 from torque for Boost 
TorqueGauge: Debug Settings max speed 0.0 
TorqueRefresher$doRefresh: Debug Got valid 0.0 from torque for HP 
TorqueRefresher$doRefresh: Debug Got valid 0.0 from torque for COM-BP-A 
TorqueRefresher$doRefresh: Debug Got valid -10.442717121373825 from torque for Boost 
TorqueGauge: Debug Settings max speed 0.0 
TorqueRefresher$doRefresh: Debug Got valid 0.0 from torque for HP 
TorqueRefresher$doRefresh: Debug Got valid 0.0 from torque for COM-BP-A 
TorqueRefresher$doRefresh: Debug Got valid -10.442717121373825 from torque for Boost 
TorqueGauge: Debug Settings max speed 0.0 
TorqueRefresher$doRefresh: Debug Got valid 0.0 from torque for HP 
TorqueRefresher$doRefresh: Debug Got valid 0.0 from torque for COM-BP-A 
TorqueRefresher$doRefresh: Debug Got valid -10.297679383576966 from torque for Boost 
TorqueGauge: Debug Settings max speed 0.0 
TorqueRefresher$doRefresh: Debug Got valid 0.0 from torque for HP 
TorqueRefresher$doRefresh: Debug Got valid 0.0 from torque for COM-BP-A 
TorqueRefresher$doRefresh: Debug Got valid -10.297679383576966 from torque for Boost 
TorqueGauge: Debug Settings max speed 0.0 
TorqueRefresher$doRefresh: Debug Got valid 0.0 from torque for HP 
TorqueRefresher$doRefresh: Debug Got valid 0.0 from torque for COM-BP-A 
TorqueRefresher$doRefresh: Debug Got valid -10.297679383576966 from torque for Boost 
TorqueGauge: Debug Settings max speed 0.0 
TorqueRefresher$doRefresh: Debug Got valid 0.0 from torque for HP 
TorqueRefresher$doRefresh: Debug Got valid 0.0 from torque for COM-BP-A 
TorqueRefresher$doRefresh: Debug Got valid -10.297679383576966 from torque for Boost 
TorqueGauge: Debug Settings max speed 0.0 
TorqueRefresher$doRefresh: Debug Got valid 0.0 from torque for HP 
TorqueRefresher$doRefresh: Debug Got valid 0.0 from torque for COM-BP-A 
TorqueRefresher$doRefresh: Debug Got valid -10.297679383576966 from torque for Boost 
TorqueGauge: Debug Settings max speed 0.0 
TorqueRefresher$doRefresh: Debug Got valid 0.0 from torque for HP 
TorqueRefresher$doRefresh: Debug Got valid 0.0 from torque for COM-BP-A 
TorqueRefresher$doRefresh: Debug Got valid -10.297679383576966 from torque for Boost 
TorqueGauge: Debug Settings max speed 0.0 
TorqueRefresher$doRefresh: Debug Got valid 0.0 from torque for HP 
TorqueRefresher$doRefresh: Debug Got valid 0.0 from torque for COM-BP-A 
TorqueRefresher$doRefresh: Debug Got valid -10.297679383576966 from torque for Boost 
TorqueGauge: Debug Settings max speed 0.0 
TorqueRefresher$doRefresh: Debug Got valid 0.0 from torque for HP 
TorqueRefresher$doRefresh: Debug Got valid 0.0 from torque for COM-BP-A 
TorqueRefresher$doRefresh: Debug Got valid -10.297679383576966 from torque for Boost 
TorqueGauge: Debug Settings max speed 0.0 
TorqueRefresher$doRefresh: Debug Got valid 0.0 from torque for HP 
TorqueRefresher$doRefresh: Debug Got valid 0.0 from torque for COM-BP-A 
TorqueRefresher$doRefresh: Debug Got valid -10.297679383576966 from torque for Boost 
[... 345 lines cut ...]
TorqueRefresher$doRefresh: Debug Got valid 0.3514739352862316 from torque for HP 
TorqueRefresher$doRefresh: Debug Got valid 0.0 from torque for COM-BP-A 
TorqueRefresher$doRefresh: Debug Got valid -10.297679383576966 from torque for Boost 
TorqueGauge: Debug Settings max speed 0.0 
TorqueRefresher$doRefresh: Debug Got valid 0.3514739352862316 from torque for HP 
TorqueRefresher$doRefresh: Debug Got valid 0.0 from torque for COM-BP-A 
TorqueRefresher$doRefresh: Debug Got valid -10.297679383576966 from torque for Boost 
TorqueGauge: Debug Settings max speed 0.0 
TorqueRefresher$doRefresh: Debug Got valid 0.3514739352862316 from torque for HP 
TorqueRefresher$doRefresh: Debug Got valid 0.0 from torque for COM-BP-A 
[... cut ...]
TorqueRefresher$doRefresh: Debug Got valid 0.0 from torque for COM-BP-A 
TorqueRefresher$doRefresh: Debug Got valid -10.297679383576966 from torque for Boost 
TorqueGauge: Debug Settings max speed 0.0 
TorqueRefresher$doRefresh: Debug Got valid 0.3514739352862316 from torque for HP 
TorqueRefresher$doRefresh: Debug Got valid 0.0 from torque for COM-BP-A 
TorqueRefresher$doRefresh: Debug Got valid -10.297679383576966 from torque for Boost 
TorqueGauge: Debug Settings max speed 0.0 
TorqueRefresher$doRefresh: Debug Got

spadab, 2023-10-17:
Copying the log was a simple copy and paste🤣. This is mine with the PID Turbo boost & Vacuum in Bar and Psi.

TorqueRefresher$doRefresh: Debug Got valid -0.14503773779685858 from torque for Boost 
TorqueGauge: Debug Settings min speed -0.5801509511874343 
TorqueRefresher$doRefresh: Debug Got valid -0.14503773779685858 from torque for Boost 
TorqueGauge: Debug Settings max speed 0.0 
TorqueRefresher$doRefresh: Debug Got valid -0.14503773779685858 from torque for Boost 
TorqueGauge: Debug Settings min speed -0.5801509511874343 
TorqueRefresher$doRefresh: Debug Got valid -0.14503773779685858 from torque for Boost 
TorqueGauge: Debug Settings max speed 0.0 
TorqueRefresher$doRefresh: Debug Got valid -0.29007547559371716 from torque for Boost 
TorqueGauge: Debug Settings min speed -0.5801509511874343 
TorqueRefresher$doRefresh: Debug Got valid -0.29007547559371716 from torque for Boost 
TorqueGauge: Debug Settings max speed 0.0 
TorqueRefresher$doRefresh: Debug Got valid -0.29007547559371716 from torque for Boost 
TorqueGauge: Debug Settings min speed -0.5801509511874343 
TorqueRefresher$doRefresh: Debug Got valid -0.29007547559371716 from torque for Boost 
TorqueGauge: Debug Settings max speed 0.0 
TorqueRefresher$doRefresh: Debug Got valid -0.29007547559371716 from torque for Boost 
TorqueGauge: Debug Settings min speed -0.5801509511874343 
TorqueRefresher$doRefresh: Debug Got valid -0.29007547559371716 from torque for Boost 
TorqueGauge: Debug Settings max speed 0.0 
TorqueRefresher$doRefresh: Debug Got valid -0.29007547559371716 from torque for Boost 
TorqueGauge: Debug Settings min speed -0.5801509511874343 
TorqueRefresher$doRefresh: Debug Got valid -0.29007547559371716 from torque for Boost 
TorqueGauge: Debug Settings max speed 0.0 
TorqueRefresher$doRefresh: Debug Got valid -0.29007547559371716 from torque for Boost 
TorqueGauge: Debug Settings min speed -0.5801509511874343 
TorqueRefresher$doRefresh: Debug Got valid -0.29007547559371716 from torque for Boost 
TorqueGauge: Debug Settings max speed 0.0 
TorqueRefresher$doRefresh: Debug Got valid -0.29007547559371716 from torque for Boost 
TorqueGauge: Debug Settings min speed -0.5801509511874343 
TorqueRefresher$doRefresh: Debug Got valid -0.29007547559371716 from torque for Boost 
TorqueGauge: Debug Settings max speed 0.0 
TorqueRefresher$doRefresh: Debug Got valid -0.29007547559371716 from torque for Boost 
TorqueGauge: Debug Settings min speed -0.5801509511874343 
TorqueRefresher$doRefresh: Debug Got valid -0.29007547559371716 from torque for Boost 
TorqueGauge: Debug Settings max speed 0.0 
TorqueRefresher$doRefresh: Debug Got valid -0.29007547559371716 from torque for Boost 
TorqueGauge: Debug Settings min speed -0.5801509511874343 
TorqueRefresher$doRefresh: Debug Got valid -0.29007547559371716 from torque for Boost 
TorqueGauge: Debug Settings max speed 0.0 
TorqueRefresher$doRefresh: Debug Got valid -0.29007547559371716 from torque for Boost 
TorqueGauge: Debug Settings min speed -0.5801509511874343 
TorqueRefresher$doRefresh: Debug Got valid -0.29007547559371716 from torque for Boost 
TorqueGauge: Debug Settings max speed 0.0 
TorqueRefresher$doRefresh: Debug Got valid -0.29007547559371716 from torque for Boost 
TorqueGauge: Debug Settings min speed -0.5801509511874343 
TorqueRefresher$doRefresh: Debug Got valid -0.29007547559371716 from torque for Boost 
TorqueGauge: Debug Settings max speed 0.0 
TorqueRefresher$doRefresh: Debug Got valid -0.29007547559371716 from torque for Boost 
TorqueGauge: Debug Settings min speed -0.5801509511874343 
TorqueRefresher$doRefresh: Debug Got valid -0.29007547559371716 from torque for Boost 
TorqueGauge: Debug Settings max speed 0.0 
TorqueRefresher$doRefresh: Debug Got valid -0.29007547559371716 from torque for Boost 
TorqueGauge: Debug Settings min speed -0.5801509511874343 
TorqueRefresher$doRefresh: Debug Got valid -0.5801509511874343 from torque for Boost 
TorqueGauge: Debug Settings max speed 0.0 
TorqueRefresher$doRefresh: Debug Got valid -0.14503773779685858 from torque for Boost 
TorqueGauge: Debug Settings min speed -0.5801509511874343 
TorqueRefresher$doRefresh: Debug Got valid -0.14503773779685858 from torque for Boost 
TorqueGauge: Debug Settings max speed 0.0 
TorqueRefresher$doRefresh: Debug Got valid 0.0 from torque for Boost 
TorqueGauge: Debug Settings min speed -0.5801509511874343 
TorqueRefresher$doRefresh: Debug Got valid 0.29007547559371716 from torque for Boost 
TorqueGauge: Debug Settings max speed 0.0 
TorqueRefresher$doRefresh: Debug Got valid 0.8702264267811515 from torque for Boost 
TorqueGauge: Debug Settings min speed -0.5801509511874343 
TorqueRefresher$doRefresh: Debug Got valid 1.1603019023748686 from torque for Boost 
TorqueGauge: Debug Settings max speed 0.0 
[... 907 lines cut ...]
TorqueRefresher$doRefresh: Debug Got valid 3.625943444921468 from torque for Boost 
TorqueGauge: Debug Settings min speed -0.5801509511874343 
TorqueRefresher$doRefresh: Debug Got valid 3.625943444921468 from torque for Boost 
TorqueGauge: Debug Settings max speed 0.0 
TorqueRefresher$doRefresh: Debug Got valid 3.1908302315308923 from torque for Boost 
TorqueGauge: Debug Settings min speed -0.5801509511874343 
TorqueRefresher$doRefresh: Debug Got valid 3.1908302315308923 from torque for Boost 
TorqueGauge: Debug Settings max speed 0.0 
TorqueRefresher$doRefresh: Debug Got valid 2.900754755937175 from torque for Boost 
TorqueGauge: Debug Settings min speed -0.5801509511874343 
[... cut ...]
TorqueGauge: Debug Settings min speed -0.5801509511874343 
TorqueRefresher$doRefresh: Debug Got valid 2.7557170181403166 from torque for Boost 
TorqueGauge: Debug Settings max speed 0.0 
TorqueRefresher$doRefresh: Debug Got valid 2.7557170181403166 from torque for Boost 
TorqueGauge: Debug Settings min speed -0.5801509511874343 
TorqueRefresher$doRefresh: Debug Got valid 2.7557170181403166 from torque for Boost 
TorqueGauge: Debug Settings max speed 0.0 
TorqueRefresher$doRefresh: Debug Got valid 2.7557170181403166 from torque for Boost

agronick, 2023-10-17:
Your boost is a negative value. Is your min value set low enough?

spadab, 2023-10-17:
> Your boost is a negative value. Is your min value set low enough?

Yes the setup is -20/20, I have already tried to modify it in various ways but it still doesn't work.  I tried decreasing the numbers after the decimal point to just two units but still nothing.

elnidalarguia, 2023-10-17:
> Tu impulso es un valor negativo. ¿Su valor mínimo está establecido lo suficientemente bajo?
In the configuration I put -2000/2000 just in case it did not register any value but nothing is not able to read anything, always marks 0 bar when I put the unit conversion from PSI to BAR.

elnidalarguia, 2023-10-17:
[image]

elnidalarguia, 2023-10-17:
[image]

elnidalarguia, 2023-10-17:
[image]

elnidalarguia, 2023-10-17:
This is my configuration

marced75, 2023-10-17:
After the release of version R5.1 unit conversions doesn't work for any kind of circular gauges. The values shown still remain to 0 also leaving only "A" in the expression. The small text indicators are not affected by this bug.  My log file is linked!
[AATorque.log](https://drive.google.com/file/d/1A3CB27lQjBELQJlg9aFUhx_Bq2ovfklJ/view?usp=drivesdk)

agronick, 2023-10-17:
Can you try just changing the formula to "a" and see if it displays anything? I don't have boost on my car but other values converted with the formula display fine.

marced75, 2023-10-17:
Same problem also with "a". Only this PID is affected by this issue.
FRESH NEWS: enabling "show only whole numbers", problem disappears. But this gauge need to have them.
The small text indicators are not affected.

elnidalarguia, 2023-10-17:
> El mismo problema también con "a". Sólo este PID se ve afectado por este problema. NUEVAS NOTICIAS: al habilitar "mostrar solo números enteros", el problema desaparece. ¡Pero este indicador necesita tener decimales! ¡Los pequeños indicadores de texto no se ven afectados! ... He solucionado el problema utilizando una conversión a "mbar" en lugar de "barra".

Can you pass a picture of how you have it, because I am trying to put it in mbar and it doesn't work.

marced75, 2023-10-17:
> Can you pass a picture of how you have it, because I am trying to put it in mbar and it doesn't work.

You need to set a custom conversion (a  / 0.0145038) to obtain values in mbar. Now you must enable "use whole numbers" to obtain correct readings.

agronick, 2023-10-17:
I think it has to do with the numbers being so long. Can you check the prerelease version and see if that fixes it?

marced75, 2023-10-18:
> I think it has to do with the numbers being so long. Can you check the prerelease version and see if that fixes it?

The letest app version fixed the problem. Thanks!

spadab, 2023-10-19:
I confirm latest app fixed the problem  too! Thank you

closed 2023-10-19 by spadab