agronick/aa-torque #40: [BUG] Custom PIDs are wrongly displayed in the AA-Torque

aivus, 2023-10-13:
**Describe the bug**

I've created 3 custom PIDs in Torque app, but AA-Torque showed only one of them duplicated 3 times.

**To Reproduce**
Steps to reproduce the behavior:
1. Create several custom PIDs in Torque app
2. In AA-Torque try to use them

**Expected behavior**
All custom PIDs are displayed in AA-Torque

**App version:** 10.0.40

**Screenshots**
[image]
[image]

agronick, 2023-10-13:
Can you make a screen recording? Is your Torque updated?

that1cardude, 2023-10-16:
I also have a few custom PIDS in Torque Pro but when I choose them in AA Torque it does not show the values. But the actual torque pro app does.

aivus, 2023-10-21:
@agronick 

> Is your Torque updated?

Latest version of Torque app.

Screen recording: https://youtube.com/shorts/5dl7g6SA_zc

aivus, 2023-10-21:
@that1cardude please create separate issue

agronick, 2023-10-25:
Can you export your pids from torque to a CSV and attach them? I think Torque is probably giving bad data. I have seen other bugs on that page in Torque.

DanSilva926, 2023-11-06:
I have a custom PID and they are working correctly, check the values ​​in the Torque app if they are correct?

aivus, 2023-11-08:
@DanSilva926 It's more about PIDs, not the data received from car.
As you can see from the screenshots and video above Torque App shows PIDs correctly in the list.

@agronick I did export from Torque, removed all custom PIDs, cleared all data of AA Torque app, imported custom PIDs to Torque. And still having this.

Attaching the exported PIDs from Torque app: [exportedPIDs.csv](https://github.com/agronick/aa-torque/files/13294205/exportedPIDs.csv)

MasterPIC, 2023-11-08:
I'm thinking about a bug related to common "Battery" string followed by space character. If this is the case then You can try to set something like Battery_xxx, Battery_yyy as PID name...

aivus, 2023-11-11:
@agronick if it's (☝️ ) a case and no issues on AA-Torque side - feel free to close the issue

agronick, 2023-11-19:
Sorry for the delay with this. I finally got a chance to look at it. The problem is that your PIDs are not in hexidecimal format. They are in decimal. The torque app might have some code to do the conversion but it is coming back through the service as 3 pids named `7ffffffe,65`. All the `f` characters look like it is encountering some kind of overflow since that is the biggest digit in hex. Convert your pid to hex and it will probably work.

closed 2023-12-29 by agronick