agronick/aa-torque #42: [BUG]

Vince56190, 2023-10-15:
When I select the preview of my configuration, the application crashes. Do you want the crash report?
Android 13 , Pixel 6a.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error
Including a logcat or Debug log is very helpful. Use Google to find out how to get those on your specific phone/Android version.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

agronick, 2023-10-16:
See if this is happening on the latest version

Vince56190, 2023-10-17:
it's ok on the last version , no crash anymore.

closed 2023-10-17 by agronick