agronick/aa-torque #46: [BUG] Versions after 10.0.33 Cause App to Crash on Samsung S21/Android 13/Toyota:  "AATorque has crashed."

Mtn-Mike, 2023-10-16:
Even with this bug, it's still an awesome app!!

**Describe the bug**
Every version after RC 6.2 (10.0.33) has resulted in the app crashing in my vehicle/phone combination (’22 Toyota 4Runner, Samsung S21, OEM head unit, Android v13).  Crash report attached below.  

I wish I could be more descriptive, but very little detail is returned by the head unit or phone when AA Torque crashes.  AATorque never opens on the head unit, and the phone just displays a toast message, "AATorque app has crashed.  Send report?"

**To Reproduce**
1. Start vehicle, and wait for AndroidAuto to load on head unit
2. Once AA is fully operational, tap "AATorque" icon in the AndroidAuto launcher/menu screen

**Expected behavior**
Normally AATorque starts right up and within about 1 second the gauges update.

**Screenshots**
None

**Smartphone**
 - Device: Samsung S21
 - OS: Android v13
 - Browser: Stock Android browser (Google)

**Additional context**
I noticed that the app started crashing as soon as development began on the preview feature (RC7).  No idea if there’s a relationship there—just what I noticed. 

I uninstalled AATorque and Torque Pro, rebooted my phone, reinstalled Torque Pro, reinstalled the latest AATorque and the crash still persists.  I rolled back to 10.0.33 and everything works great!  I'm good with the old version, but if you add a feature--like warning limits--I'll want to upgrade, so I'm hoping this bug is eradicated. :)

Thanks for a great app!

**Crash Report**
REPORT_ID=b4307e49-de7a-48a2-a20c-165426cc1289
APP_VERSION_CODE=1053
APP_VERSION_NAME=10.0.53
PACKAGE_NAME=com.aatorque.stats
FILE_PATH=/data/user/0/com.aatorque.stats/files
PHONE_MODEL=SM-G991U
BRAND=samsung
PRODUCT=o1qsqw
ANDROID_VERSION=13
BUILD=BOARD=lahaina
  BOOTLOADER=G991USQS9EWI2
  BRAND=samsung
  CPU_ABI=arm64-v8a
  CPU_ABI2=
  DEVICE=o1q
  DISPLAY=TP1A.220624.014.G991USQS9EWI2
  FINGERPRINT=samsung/o1qsqw/o1q:13/TP1A.220624.014/G991USQS9EWI2:user/release-keys
  HARDWARE=qcom
  HOST=SWDK6111
  ID=TP1A.220624.014
  IS_DEBUGGABLE=false
  IS_EMULATOR=false
  MANUFACTURER=samsung
  MODEL=SM-G991U
  ODM_SKU=unknown
  PERMISSIONS_REVIEW_REQUIRED=true
  PRODUCT=o1qsqw
  RADIO=unknown
  SKU=unknown
  SOC_MANUFACTURER=QTI
  SOC_MODEL=SM8350
  SUPPORTED_32_BIT_ABIS=["armeabi-v7a","armeabi"]
  SUPPORTED_64_BIT_ABIS=["arm64-v8a"]
  SUPPORTED_ABIS=["arm64-v8a","armeabi-v7a","armeabi"]
  TAGS=release-keys
  TIME=1695359049000
  TYPE=user
  UNKNOWN=unknown
  USER=dpi
  VERSION.ACTIVE_CODENAMES=[]
[... 530 lines cut ...]
  glEsVersion=3.2
ENVIRONMENT=getDataDirectory=/data
  getDataSystemDirectory=/data/system
  getDownloadCacheDirectory=/data/cache
  getExternalStorageDirectory=/storage/emulated/0
  getExternalStorageState=mounted
  getInternalMediaDirectories=[/system/media, /oem/media, /product/media, /prism/media/carriers/VZW, /data/overlays/media]
  getLegacyExternalStorageDirectory=/sdcard
  getOdmDirectory=/odm
  getOemDirectory=/oem
  getProductDirectory=/product
  getProductServicesDirectory=/product_services
  getRootDirectory=/system
  getStorageDirectory=/storage
  getSystemExtDirectory=/system_ext
  getVendorDirectory=/vendor
  isExternalStorageEmulated=true
  isExternalStorageLegacy=false
  isExternalStorageManager=false
  isExternalStorageRemovable=false
SHARED_PREFERENCES=default.selectedBackground=background_incar_black
  default.rotaryInput=false
  default.selectedFont=frutiger
  default.centerGaugeLarge=false
  default.selectedTheme=Seat Cupra

agronick, 2023-10-16:
Let me know if this is fixed in RC10

Mtn-Mike, 2023-10-16:
> DuckDuckGo removed one tracker. More
>
> Report Spam

Hey Kyle,

 

I posted an update to the Issues section on GitHub.  Take a quick look.

 

TL;DR:  App no longer crashes, but the display went a little wonky.

 

 

https://github.com/agronick/aa-torque/issues/46

 

-Mike

agronick, 2023-10-16:
Yeah one of the changes was to undo the hardcoded dimensions the previous developer had put. Now it scales better especially on larger screens. I noticed the high visibility dial moved but I wasn't too bothered by it. It would look better if it was back where it was though.

closed 2023-10-16 by agronick

Mtn-Mike, 2023-10-16:
Ah.  Making it compatible with a wider variety of screens (scaling) is a worthy update, for sure.  If you decide to keep tweaking the scaling, I'll be happy to test it--I'd love to see it back where it was in 10.0.33, but I know this isn't your life's work.  I appreciate your continued development, but I'm gonna stick with 10.0.33 for now--in my use case, it's dang-near perfect.

This may not be the right place to ask, but can the user change the ray color?  I know it's not a menu item...was just wondering if there's a file or a bit in the APK (or something) that can be twiddled to make the ray blue.  (Toyota uses blue for all its displays, switchgear, etc.)