agronick/aa-torque #47: [REQUEST] add a way to import custom assets

articrevised, 2023-10-16:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
I think it would be awesome to add a sort of color picker for some of the preset gauges. Or just a way to import custom assets like fonts, gauges, and backgrounds

If there is a easy way to do that already in android studio please let me know

articrevised, 2023-10-18:
> It's doable. Anything besides colors would take a bit of work. If you edit styles.xml you can configure most of the layout. It is all in the resource files.

Ok cool! I'll give it a try

agronick, 2023-11-01:
Check out this new guide for how to add themes. Its all in xml now. https://github.com/agronick/aa-torque/pull/63

agronick, 2023-11-03:
Fonts are controlled by the 3 arrays defined after this line: https://github.com/agronick/aa-torque/blob/9f72f20e109228938079b01cdf12340b3fa32c36/app/src/main/res/values/arrays.xml#L82

Backgrounds are defined at the 3 arrays defined after this line: https://github.com/agronick/aa-torque/blob/9f72f20e109228938079b01cdf12340b3fa32c36/app/src/main/res/values/arrays.xml#L271

Strings need to be defined in each language. I just do the English ones and wait for people to fill them in on Poeeditor.

mkwatchface, 2023-11-03:
Thank you! I will try to play with it locally!

mkwatchface, 2023-11-04:
> Fonts are controlled by the 3 arrays defined after this line:
> 
> https://github.com/agronick/aa-torque/blob/9f72f20e109228938079b01cdf12340b3fa32c36/app/src/main/res/values/arrays.xml#L82
> 
> Backgrounds are defined at the 3 arrays defined after this line:
> 
> https://github.com/agronick/aa-torque/blob/9f72f20e109228938079b01cdf12340b3fa32c36/app/src/main/res/values/arrays.xml#L271
> 
> Strings in these arrays need to be defined in each language in the strings.xml file. I just do the English ones and wait for people to fill them in on Poeeditor.

Hi Kyle,

Playing a bit with the above information i managed to add a background and of course a new theme. I am having a bit of a problem with the font though. I added the 3 arrays in the strings and now the font thumb appears in the list of fonts but when i apply it, it does not change. so i added the following:

`    <string-array name="fontEntries">
	<item>@string/font_ev</item>
    </string-array>`

Then:

`    <string-array name="fontValues">
	<item>ev</item>
    </string-array>`

And finally for the thumb:

`    <string-array name="FontThumbs">
	<item>@drawable/ic_font_ev</item>
    </string-array>`

What i miss is the link to the actual "Font" folder. The font that i placed in that folder is called ev.otf

Looking into the existing fonts you have there, i can find otf fonts so this should be compatible, however, i also looked into the rest of the code in the strings.xml and there is no link to the font folder.

Sorry for the noob questions :)

agronick, 2023-11-04:
That looks awesome. I forgot the font had some code that isn't in XML here. https://github.com/agronick/aa-torque/blob/9f72f20e109228938079b01cdf12340b3fa32c36/app/src/main/java/com/aatorque/stats/DashboardFragment.kt#L293 It could probably be converted to an XML array but the code is pretty easy. Just map what is in `fontValues` to `R.font.whatever`.

mkwatchface, 2023-11-05:
> That looks awesome. I forgot the font had some code that isn't in XML here: [...]

That's great! It worked! i have the font now working as well... I will test for a couple of days and if all is fine i will create a pull request. I will try to build again because everytime i try to force update, it doesn't installed giving me an invalid package error but your stable release installs fine.

I have also tried with King Installer and i have the same error.

Will keep trying!
Kind Regards

mkwatchface, 2023-11-05:
First test looks good!!!

[image]

agronick, 2023-11-06:
That's awesome. I think people will love it. Doesn't look like any of the other themes.

mkwatchface, 2023-11-06:
> That's awesome. I think people will love it. Doesn't look like any of the other themes.

Yeah, it looks a bit more modern than the current themes. Personally, I love it even though it is not based on any actual model, for those with blue dials that can match it it will be another option.