agronick/aa-torque #50: [BUG] Engine Oil temperature not work

elnidalarguia, 2023-10-18:
The engine oil temperature is not displayed, always marks the minimum temperature that you mark in the app to be displayed, I tried putting without decimals and it does not work either.

elnidalarguia, 2023-10-18:
[image]

DanSilva926, 2023-10-20:
[image]
 Oil work temp

elnidalarguia, 2023-10-20:
Ok it's Torque's fault that it doesn't show me the temperature. How can I create a custom pid.

DanSilva926, 2023-10-20:
In the torque settings there are extra pid options, you can add several brands

closed 2023-10-20 by agronick