agronick/aa-torque #51: [BUG] dont have options for remove background

DanSilva926, 2023-10-20:
[image]

I selected a background to see if I liked one but now I can't remove it and I'm left with no background.

agronick, 2023-10-22:
Black is remove background

DanSilva926, 2023-10-22:
> Preto é remover fundo

No ,is diferent .
the black one has a line on top of the pressure gauges, I'll leave a photo later

[image]

DanSilva926, 2023-10-23:
[image]

if you choose black, that line appears at the top

commit d1060360 (2023-10-25): fix: always provide a background fixes #51
closed 2023-10-25 by agronick