agronick/aa-torque #55: Boost/ Vaccum PSI to Bar issue

Vince56190, 2023-10-22:
| :exclamation:  Issues that aren't in English will  be closed and igored.  |
|-----------------------------------------|
Still the problem of converting the Vacuum / Boost gauge when converting to bar. **It is noted that this value can be negative (Vaccum)**

**Describe the bug**
A clear and concise description of what the bug is.
Be as descriptive as possible, and write in English.
Make sure you've read the Readme and searched for similar issues before asking a question.
Issues not in English or without any detail will not be closed and ignored. 
Be kind and respectful and maybe most importantly: Don't be lazy (just posting a duplicate error like "it's crashing" or "it doesn't get data") while you're expecting me to do all the work for free 😄

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error
Including a logcat or Debug log is very helpful. Use Google to find out how to get those on your specific phone/Android version.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

elnidalarguia, 2023-10-22:
> ❗ Los problemas que no estén en inglés se cerrarán y se ignorarán.
> Sigue existiendo el problema de convertir el indicador de vacío/impulso al convertirlo a bar. **Se observa que este valor puede ser negativo (Vaccum)**
> **Describe el error** Una descripción clara y concisa de qué es el error. Sea lo más descriptivo posible y escriba en inglés. Asegúrese de haber leído el archivo Léame y haber buscado problemas similares antes de hacer una pregunta. Los problemas que no estén en inglés o que no contengan ningún detalle no se cerrarán ni se ignorarán. Sea amable y respetuoso y quizás lo más importante: no sea perezoso (simplemente publique un error duplicado como "está fallando" o "no obtiene datos") mientras espera que yo haga todo el trabajo gratis 😄
> 
> **Para reproducir** Pasos para reproducir el comportamiento:
> 
> 1. Ir a '...'
> 2. Haga clic en '....'
> 3. Desplácese hacia abajo hasta '....'
> 4. Ver error 
>    Incluir un logcat o un registro de depuración es muy útil. Utilice Google para descubrir cómo obtenerlos en su teléfono/versión de Android específica.
> 
> **Comportamiento esperado** Una descripción clara y concisa de lo que esperaba que sucediera.
> 
> **Capturas de pantalla** Si corresponde, agregue capturas de pantalla para ayudar a explicar su problema.
> 
> **Escritorio (por favor complete la siguiente información):**
> 
> * SO: [por ejemplo, iOS]
> * Navegador [por ejemplo, Chrome, Safari]
> * Versión [por ejemplo, 22]
> 
> **Smartphone (por favor complete la siguiente información):**
> 
> * Dispositivo: [por ejemplo, iPhone6]
> * SO: [por ejemplo, iOS8.1]
> * Navegador [por ejemplo, navegador de valores, safari]
> * Versión [por ejemplo, 22]
> 
> **Contexto adicional** Agregue aquí cualquier otro contexto sobre el problema.

Hello friend, the values are displayed correctly, if you see vacuum in your gauge is because it measures in relative pressure and when the throttle is closed you see negative value because there is vacuum in the intake manifold.
Traducido con DeepL https://www.deepl.com/app/?utm_source=android&utm_medium=app&utm_campaign=share-translation

Vince56190, 2023-10-22:
it is not a vacuum problem, it works very well in PSI but in Bar it does not work at all, neither negative nor positive

elnidalarguia, 2023-10-22:
Strange, it works perfectly for me

Vince56190, 2023-10-22:
I don't no why but now it work  , i just change Lustang to Audi cockpit.  , you can close this issue . Thx

closed 2023-10-22 by agronick