agronick/aa-torque #56: [BUG]oil temperature

Fabio1968, 2023-10-25:
Hi, congratulations on the app, it's very well done.  My oil temperature gauge doesn't work.  does this happen to you too?  

> thank you for your response

 :exclamation:  Issues that aren't in English will  be closed and igored.  |
|-----------------------------------------|

**Describe the bug**
A clear and concise description of what the bug is.
Be as descriptive as possible, and write in English.
Make sure you've read the Readme and searched for similar issues before asking a question.
Issues not in English or without any detail will not be closed and ignored. 
Be kind and respectful and maybe most importantly: Don't be lazy (just posting a duplicate error like "it's crashing" or "it doesn't get data") while you're expecting me to do all the work for free 😄

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error
Including a logcat or Debug log is very helpful. Use Google to find out how to get those on your specific phone/Android version.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

agronick, 2023-10-25:
Is oil temperature reported in torque?

Fabio1968, 2023-10-26:
Thanks for the reply.  I had assumed that the sensor was detected by torque and consequently by aa-torque and instead it doesn't work

agronick, 2023-11-01:
Yeah, its just asking torque for a list of PIDs. It isn't checking that they all work, that would take too long. Maybe eventually I'll have it report live data in the list like Torque does.

closed 2023-11-01 by agronick