agronick/aa-torque #57: [BUG] Headlight/illumination toggle from car causes app reset on radio display

AltyFox, 2023-10-25:
**Describe the bug**
When toggling the headlights/ the illumination line on the radio, the app on the screen completely resets. For certain PIDs like calculated trip MPG, resets all calculations.

**To Reproduce**
Toggle illumination
**Expected behavior**
The app doesn't reset .

**Smartphone (please complete the following information):**
Pixel 6 pro Android 14

Vince56190, 2023-10-25:
off topic: Does it work with pixel 6 and android 14? what procedure did you use to install it? At first it didn't work. THANKS

closed 2023-10-25 by agronick

AltyFox, 2023-10-25:
> off topic: Does it work with pixel 6 and android 14? what procedure did you use to install it? At first it didn't work. THANKS

I'm rooted, and enabled developer mode on Android auto as well as my AA wireless dongle.

commit 9eaf6416 (2023-10-25): fix: activity recreate on night mode (fixes #57)