agronick/aa-torque #61: [BUG] Crash on Android 14

aross6, 2023-10-30:
Android 14
Pixel 6a
AA Torque 12.2

App worked fine on Android 13. Once phone was updated to Android 14, AA Torque crashes when launched from Android Auto. Crash Report sent via Gmail.

Vince56190, 2023-10-31:
if no rooted this app does not work on Android 14

aross6, 2023-10-31:
I think this may be a separate issue than you're thinking. AA Torque shows
in Android Auto with my phone on Android 14 because I sideloaded the app
while still on Android 13. So it shows in the Android Auto Launcher on my
radio(s) just fine. But something is causing it to crash when trying to
launch it from Android Auto with my phone on Android 14, which it did not
do on Android 13.

Dissimilar, Fermata also shows in Android Auto with my phone on Android 14
because I also sideloaded it while still on Android 13. So it also shows in
the Android Auto Launcher on my radio(s) just fine, but it is still able to
open/launch/work from Android Auto with my phone on Android 14.

This leads me to believe that it's a Big with AA Torque making unable to be
launched from Android Auto with my phone on Android 14.

Vince56190, 2023-10-31:
> Je pense que cela peut être un problème différent de celui que vous pensez. AA Torque s'affiche dans Android Auto avec mon téléphone sous Android 14, car j'ai téléchargé l'application alors que j'étais encore sous Android 13. Il s'affiche donc très bien dans le lanceur Android Auto sur ma (mes) radio(s). Mais quelque chose le fait planter lorsque j'essaie de le lancer depuis Android Auto avec mon téléphone sous Android 14, ce qu'il n'a pas fait sur Android 13. Contrairement à lui, Fermata s'affiche également dans Android Auto avec mon téléphone sous Android 14 car je l'ai également chargé de côté. tout en étant toujours sous Android 13. Il s'affiche donc également très bien dans le lanceur Android Auto sur mes radios, mais il est toujours capable d'ouvrir/lancer/travailler depuis Android Auto avec mon téléphone sous Android 14. Cela m'amène à croire que c'est un Big avec AA Torque qui ne peut pas être lancé depuis Android Auto avec mon téléphone sous Android 14.
> […](#)

BUt if you update AA Torque , the App will not be in AA Anymore.

agronick, 2023-10-31:
The bug report you sent me showed that it was complaining that a file was writable. It looks like a new security check in Android 14. I updated the app with some code I found from stack overflow but I'm not sure if it will fix it. The easiest way to fix it will be to uninstall and reinstall.

agronick, 2023-10-31:
Also people have gotten it working on Android 14 so I think a reinstall will get it working the Android 14 way. https://github.com/agronick/aa-torque/discussions/31

aross6, 2023-11-02:
> Also people have gotten it working on Android 14 so I think a reinstall will get it working the Android 14 way. #31

Is the updated code that you feel should resolve the app crashing in the newest version (14.4)?

agronick, 2023-11-06:
15.1

agronick, 2023-11-21:
Should be resolved

closed 2023-11-21 by agronick