agronick/aa-torque #62: [BUG] when moving to another app such as Google Maps, AA torque loses some values from trip

omaralwardian, 2023-10-31:
| :exclamation:  Issues that aren't in English will  be closed and igored.  |
|-----------------------------------------|

Describe the bug 
Specifically related to fuel used. There seems to be a bug where no matter if the phone has Torque Pro battery unrestricted with all permissions and connected to the OBD2 adapter, or AA Torque with the same settings except for the connection to the adapter, the fuel used amount always resets back to zero when leaving and returning to the Torque dashboard within the same drive.

This doesn’t happen with all other PIDs I have activated on the dashboard, such as kilometers, trip time, trip distance, etc.

To Reproduce 
Create a dial with fuel used and another with trip time. Ensure you are fully connected, start driving, switch off the screen or move to a different application such as Google Maps. Then, go back to AA Torque, and you will see that the fuel used level reverts back to zero. Meanwhile, kilometers driven since journey started will remain accurate.

Expected behavior 
The fuel used should continue to tally even when backgrounded.

Smartphone 
Samsung Galaxy S21 Ultra Snapdragon (North American variant) 
Android 13

MasterPIC, 2023-11-02:
Could (internal) data logging by Torque solve the issue?
You can add PIDs to be logged and, as for trip recording, you can set Torque to ignore short gaps between trips.
I don't know if polling logged PIDs will force Torque to return logged data to AA-Torque.
Unfortunately I'm currently not able to make any test.

agronick, 2023-11-03:
From what I've observed with how Toque works I would say it probably would.

agronick, 2023-11-19:
Did that work? It might be a workaround for https://github.com/agronick/aa-torque/issues/73 as well.

agronick, 2024-01-04:
This hasn't been updated in over a month. Without a response it will be closed.

closed 2024-01-08 by agronick