agronick/aa-torque #68: [REQUEST] User-selectable gauge needle/ray color

Mtn-Mike, 2023-11-02:
**Describe the solution you'd like**
Currently, gauge needle/ray color is hardcoded in each theme.  Leverage the new color picker function you've implemented for chart line colors and use it to select a custom color for gauge needles/rays.  I'm not saying to use the **_same_** color chosen for a chart line (though, that would be an option to consider), just use the color picker module you've coded and use it to let the user set the needle color variable (overriding the hard-coded, theme-based color).

**Describe alternatives you've considered**
No real alternative.  (I installed the Android SDK, fiddled with the code, and got the color I wanted, but I broke other stuff. =8-D

**Additional context**
None.

agronick, 2023-11-19:
Yeah this would be pretty easy. Maybe I'll get to it in a week or two.

Mtn-Mike, 2023-11-19:
This will be awesome.  I appreciate you considering it.  Thanks!