agronick/aa-torque #70: LineageOS 20 with mircog not responsive

delmocro, 2023-11-05:
Hello, thanks for making Torque available on AA. Ik have installed everything the right way and AA Torque shows up as on the headunit. The problem i have is that the AA Torque application on the phone is unresponsive. None of the buttons react to input. I tried doing this multiple times while also having Torque pro working in the background. Just doesnt respond to input.  I also uninstalled and installed multiple times with kinginstaller and just cant get is to respond. Does this have anything  to do with MicroG?  3 dots also not responding.

delmocro, 2023-11-12:
> Hello, thanks for making Torque available on AA. Ik have installed everything the right way and AA Torque shows up as on the headunit. The problem i have is that the AA Torque application on the [...]

Nobody have the same problem?

agronick, 2023-11-19:
That is strange. I don't know how I could reproduce it though. It is like it is blocking but I don't know what it would be blocking on. Do you know how to hook up a debugger?

delmocro, 2023-11-19:
> That is strange. I don't know how I could reproduce it though. It is like it is blocking but I don't know what it would be blocking on. Do you know how to hook up a debugger?

Unfortunatly i dont. I have tried on another phone samsung s8 also with lineageos microg and same thing. Altough on that phone if you keep tapping the dialog appears but its intermittent (you really have to catch it) so i dont know whats blocking it. I have AFwall+ ant allowed internet connection to the app. 
The app shows as installed from microG Companion. So not from Google play store but should be the same i guess.

agronick, 2023-11-20:
The dialog for Android Auto permissions?

delmocro, 2023-11-20:
> The dialog for Android Auto permissions?

No i mean when you press on the 3 dots in aa torque. 

I wil try to hook up a debugger. Yesterday i tried on android studio but couldnt get my phone attached to the debugger so as soon as i get that to work i will try to catch the problem :)

delmocro, 2023-11-22:
> That is strange. I don't know how I could reproduce it though. It is like it is blocking but I don't know what it would be blocking on. Do you know how to hook up a debugger?

Here is the logcat

Fatal signal 11 (SIGSEGV), code 2 (SEGV_ACCERR), fault addr 0x7e028850e8 in tid 10837 (JDWP Transport ), pid 10819 (.aatorque.stats)
FORTIFY: pthread_mutex_lock called on a destroyed mutex (0x7ddeb8a148)
FORTIFY: pthread_mutex_lock called on a destroyed mutex (0x7e01831cd8)
FORTIFY: pthread_mutex_lock called on a destroyed mutex (0x7e01831cd8)
Debugger has connected
waiting for debugger to settle...

debugger has settled (1392)
Unable to open '/data/app/~~6541ar3Rcw0Jc7Ptsqb2Qw==/com.aatorque.stats-_FK_msE1lhz2sSyWgJuoog==/base.dm': No such file or directory
Unable to open '/data/app/~~6541ar3Rcw0Jc7Ptsqb2Qw==/com.aatorque.stats-_FK_msE1lhz2sSyWgJuoog==/base.dm': No such file or directory
ANGLE Developer option for 'com.aatorque.stats' set to: 'default'
ANGLE GameManagerService for com.aatorque.stats: false
Neither updatable production driver nor prerelease driver is supported.
No Network Security Config specified, using platform default
No Network Security Config specified, using platform default
ACRA is enabled for com.aatorque.stats, initializing...
Verification of boolean com.aatorque.prefs.SettingsActivity.onOptionsItemSelected(android.view.MenuItem) took 121.743ms (1010.32 bytecodes/s) (4504B approximate peak alloc)
Checking for metadata for AppLocalesMetadataHolderService : Service not found
Accessing hidden method Landroid/view/View;->computeFitSystemWindows(Landroid/graphics/Rect;Landroid/graphics/Rect;)Z (unsupported, reflection, allowed)
Accessing hidden method Landroid/view/ViewGroup;->makeOptionalFitsSystemWindows()V (unsupported, reflection, allowed)
Compat change id reported: 210923482; UID 10282; state: ENABLED
Checking for update
Compat change id reported: 237531167; UID 10282; state: DISABLED
Expecting binder but got null!
Skipped 35 frames!  The application may be doing too much work on its main thread.
tagSocket(89) with statsTag=0xffffffff, statsUid=-1
Build Date                       : 07/26/21
OpenGL ES Shader Compiler Version: EV031.35.01.09
Local Branch                     : 
Remote Branch                    : 
Remote Branch                    : 
Reconstruct Branch               : 
Build Config                     : S P 10.0.7 AArch64
PFP: 0x016dc094, ME: 0x00000000
Unable to match the desired swap behavior.
Adding additional valid usage bits: 0x8202000
Compat change id reported: 78294732; UID 10282; state: ENABLED
Installing profile for com.aatorque.stats
A resource failed to call close.

agronick, 2023-11-22:
Nothing really stands out. Try setting a breakpoint in the onCreate method of SettingsActivity and step through it to see where it gets stuck. Do you have some kind of firewall? It checks for updates on github but it happens off the main thread.

delmocro, 2023-11-24:
> Nothing really stands out. Try setting a breakpoint in the onCreate method of SettingsActivity and step through it to see where it gets stuck. Do you have some kind of firewall? It checks for updates on github but it happens off the main thread.

I did that and i couldnt find the issue. I also removed the line that checks for updates checkupdate() and ran the app and still the same issue. 

I am using Afwall+  but i disabled it to see if the app works but stil same thing. 

I will try to find where the problem is, but i really dont know where to look.

agronick, 2023-12-15:
Hit the pause button when the settings screen freezes. It should tell you where it is stuck. If it is somewhere deep inside the Android SDK go up the call graph until you see something in SettingsActivity or SettingsFragment. Or you can make a breakpoint and step line by line til it freezes.

zovallo, 2023-12-18:
I have the same issue. Aa torque settings screen freezes.

zovallo, 2023-12-20:
> Nobody have the same problem?

I have your samw exact problem on my lineage os pixel 7pro phone

agronick, 2023-12-20:
If one of you can set up Android studio I would be willing to do a remote access session. Or someone could ship me a phone to test on.

zovallo, 2023-12-21:
Dear Kyle

thank for your answer

I have never set up android studio, but after xmas I will have a go.

I m pretty sure there is something in Lineage OS for Micro G that freezes the screen settings.

Also nothing from troque appears on Android auto although the app is shown and opens in AA.

Please note that on Lineage is installed AA for Micro G:

[GitHub - sn-00-x/aa4mg: Android Auto 4 microG](https://github.com/sn-00-x/aa4mg)

regards

mb

agronick, 2024-01-06:
I spent a lot of time on this. I tried to compile it from source but it is massive and couldn't get it to compile right. I found a VM but it was too old. If you want to set up a Zoom call and see if we can fix it let me know. They don't make a rom for my phone.

zovallo, 2024-01-06:
Yes sure I want. Can we do tomorrow afternoon or monday?
Thank you for the effort i think it is important for developing the app
Regards
Mb

agronick, 2024-01-06:
Look me up on facebook. We can schedule something there. I am in Eastern Standard Time.

zovallo, 2024-01-07:
Hi mr. Agronick
I have no facebook nor i want to use it, as also i do not want to have any google service. This for privacy reasons.
I can use zoom if you like. Just suggest me when. Usually a good time is in the evenings of saturday and sundays ore lunchtime on working days. Please note i live in UST +1
Regards
Mb

agronick, 2024-01-07:
I just didn't want to put my email out in the open and github removed direct messaging. I'll just post the link here. Does 18:00 UTC+1 work for you?

zovallo, 2024-01-07:
for me is ok 18:00 UTC+1!

please give me the link as i do not see it

agronick, 2024-01-08:
Yeah I didn't hear back from you in time. We can schedule something this week.