agronick/aa-torque #73: Speed (GPS) - not always working

DerGue69, 2023-11-08:
I added Speed (GPS) to see real GPS speed.
But when I launch AA Torque it gets no data.

When AA Torque is running on the Car - And I lauch the App on the phone and go to the Speed (GPS) PID - Then it get's displayed.

[image]

[image]

agronick, 2023-11-19:
It seems like a bug in Torque. Does it do that if you have torque open on your phone?

DerGue69, 2023-11-21:
I have 3 different states:

**Plug phone to car - AA Torque running on car only:**
Here I have the described issue: All data, except GPS data, working on Car display.

**Plug Phone to car - AA Torque running on car - start Torque Pro App:**
Only GPS Data is working on AA Torque. Other data freeze.

**Plug Phone to car - AA Torque running on car - AA Torque app running on phone - Again pick "Speed (GPS)" PID:**
Only then all data + gps data are working.

> It seems like a bug in Torque. Does it do that if you have torque open on your phone?

agronick, 2023-11-21:
Try setting data logging for GPS speed in Torque so it is always refreshing it.

DerGue69, 2023-11-23:
> Try setting data logging for GPS speed in Torque so it is always refreshing it.

Automatically log GPS was already activated in Torque:
[image]

Addedd "Speed (GPS) + GPS Altitude" in the "Logging preferences"
[image]

Still same: No Speed (GPS) on the Dashboard. (sorry for blurry immage...)
[image]

agronick, 2023-11-24:
See my comment [here](https://github.com/agronick/aa-torque/discussions/80#discussioncomment-7662594).