agronick/aa-torque #77: [REQUEST]Turbo pressure

Mika91100, 2023-11-14:
Hello first of all thank you for this magnificent application I wanted to report a small problem that I am encountering with the turbo pressure.  I have a turbo that takes a little more than 3 bar and when I set 3.5 bar in the application once on the road it never exceeds 1.6 bar.  Can you please make something that has greater reading value.  Thanks in advance .

agronick, 2023-11-19:
Change the max value. The app doesn't know it is displaying turbo pressure. It just takes numbers from torque and puts them on the screen.

Mika91100, 2023-11-20:
Hello thank you for your response in fact I have already modified the value to 4 bar but the reading is up to 1.5 1.6.

agronick, 2023-11-21:
Do you see the values reported correctly in Torque? The app doesn't know what value is what. It is just asking Torque for data.

closed 2024-01-03 by agronick