agronick/aa-torque #78: [BUG] 10.0.93 - center dial on dashboards, high/low value below dial.

omaralwardian, 2023-11-19:
10.0.93 appears to have a small issue with the center dial. The dial works fine but if you have a high or low value set to appear below it, it is cut off at the bottom of the screen. This issue wasn't present in earlier versions. I checked on the forums to see if I could find it already being reported somewhere and I could not. It was not in the discussions either.

Phone is the Samsung Galaxy S21 Ultra Snapdragon North American variant 888 on a Honda Civic 2018 Touring Edition infotainment screen. Running Android 13, One UI 5.1.

[image]

agronick, 2023-11-21:
I moved it below the guage. Since the widths and heights of the guage are no longer hardcoded it makes it harder to put it in the right spot. Let me know if it isn't fixed in the next version.

closed 2023-11-21 by agronick

omaralwardian, 2023-11-23:
Issue is resolved.