agronick/aa-torque #82: [BUG] Misplaced graduations on 16.5 and invalid package for previous version

Vince56190, 2023-11-25:
With the new version, the gauge values are too close to the center, they were good on the old versions. you can see that on the preview on your phone .Unfortunately I can no longer install older versions, I receive an invalid package message (installation with Kinginstaller)

agronick, 2023-12-01:
I think the biggest issue is just that font tends to be larger than the other fonts. I can try to shrink it.

Vince56190, 2023-12-06:
the problem is always present, whatever the font
[image]

Vince56190, 2023-12-10:
before it was like that : 
[image]

agronick, 2023-12-14:
Setting will be added in the next version which will be released in a few minutes

closed 2023-12-14 by agronick