agronick/aa-torque #86: [BUG] keeps crashing on Android auto

Tundra1226, 2023-12-05:
| :exclamation:  Please write your issue in English  | While trying to use this app on Android Auto it just starts crashing, and it won't open up since the last update.
|-----------------------------------------|

**Describe the bug**
Check the discussions page if this is not a bug. A clear and concise description of what the bug is.
Be as descriptive as possible. Check the [discussions](https://github.com/agronick/aa-torque/discussions)
page first. If this is not a bug, you can make a post on the discussions page.

**Log output**
After your issue happens click the three dots on the settings menu to obtain the log output and 
paste it here. If the application is crashing, you can copy and paste the email it generates.

**Media**
Capture screenshots or take a video of what is happening.

**Phone**
Leave information on the phone, car, or head unit this issue is happening on.

agronick, 2023-12-13:
Did you send the email when it crashes?

Tundra1226, 2023-12-13:
No because it only crashes when I try to open it up on my Android auto but
it doesn't crash when I open it up on my phone.

agronick, 2023-12-14:
If it is open at the same time you will get a prompt for the email.

agronick, 2024-01-04:
Please provide log output or this issue will be closed.

closed 2024-01-17 by agronick