agronick/aa-torque #87: [BUG] Updated Android Auto - Now AA Torque is not visible

DerGue69, 2023-12-09:
Updated Galaxy S22 to Android 14
Android Auto updated to 10.0.98

Now AA Torque is not longer there on the car.

> Tried reboot
> Reinstalled AA T. Reinstalled via King installer.
> Deleted Android Auto Cache and data...rebooted again. Reinstalled AAT again

Still not here.

lm1970, 2023-12-09:
Same for me...

Vince56190, 2023-12-11:
This app not work on Android 14

DerGue69, 2023-12-11:
Can you make it work on Android 14?

lm1970, 2023-12-13:
Hi agronick :) With Android 14 the app worked. It is after updating the app to version 10.0.98 that it stopped working on the car's display.

agronick, 2023-12-13:
If you downgrade it works fine?

lm1970, 2023-12-13:
I've tried with various versions just now and unfortunately it doesn't work. But I'm sure I installed Android 14 on December 4th and updated the app on the 8th...

DerGue69, 2023-12-13:
This was the version of Android Auto, whete AA Torque stopped working:

[image]

agronick, 2023-12-14:
Sounds like Google are being dicks. Not much I can do.

lm1970, 2023-12-14:
Thanks anyway for your work and I will make another donation if it starts working again. I've been using AAWireless dongle for at least a year... But the AA Torque app no longer appears in the list of Android Auto applications on my smartphone.

DerGue69, 2023-12-14:
Can you describe in technical detail, why it is not working anymore?
Since other apps still work on Android Auto : What is their officall way to become an AA app? 
And why can't you do the official way?

agronick, 2023-12-14:
There is no official way to work outside of their templates afaik. Their templates are limited to putting items in a list or a grid.

lm1970, 2023-12-15:
Usando il dongle AAWireless e abilitando la modalità sviluppatore nello stesso, ha ripreso a funzionare! 
Using the AAWireless dongle and enabling developer mode on it, it started working again! 😃

agronick, 2023-12-17:
> Thanks anyway for your work and I will make another donation if it starts working again. I've been using AAWireless dongle for at least a year... But the AA Torque app no longer appears in the list of Android Auto applications on my smartphone.

Try enabling developer mode in AAWireless

1cmoody23, 2023-12-25:
I am AA 11.063514 and android 13, its working for me. Could one of you guys try clearing cache from AA after installing AAtorque and then reboot? Just curious if that helps. Thats what I had to do to get it to show up in the car's display

kacnje, 2023-12-26:
For some weeks I have used one of the versions from November (?). Sorry, I don't know which version. But it worked on Android 14 without any problems. 
Yesterday I installed AA Torque 10.0.105 and it suddenly doesn't show in AA. 
Then I noticed that "Force update" downloaded even a newer version, but version 10.1.5 can't be installed. I get the message: "App not installed as package appears to be invalid".

kacnje, 2023-12-26:
> I am AA 11.063514 and android 13, its working for me. Could one of you guys try clearing cache from AA after installing AAtorque and then reboot? Just curious if that helps. Thats what I had to do to get it to show up in the car's display

I tried this, but it didn't help.

1cmoody23, 2023-12-26:
I got that same error, dont know whats going on with it?
Thanks,

agronick, 2023-12-26:
Try again. There was a versioning issue.

kacnje, 2023-12-27:
@agronick - thank you for your quick response. 
Installation was successful, and now I have version 2.0.2 on my phone (Android 14). 
But - I'm writing this with sadness - AA Torque still isn't available on AA. It worked on Android 14 some days ago when I had an older version.

Ekosjob, 2023-12-31:
> Updated Galaxy S22 to Android 14 Android Auto updated to 10.0.98
> 
> Now AA Torque is not longer there on the car.
> 
> * Tried reboot
> * Reinstalled AA T. Reinstalled via King installer.
> * Deleted Android Auto Cache and data...rebooted again. Reinstalled AAT again
> 
> Still not here.

Hi, any news? I've the same issue on s23 Android 14 and 1.9.63

1cmoody23, 2023-12-31:
Have you put android auto into developer mode? I would try that and then clear cache again.

Ekosjob, 2024-01-02:
hi, yes it's not working

tuononh, 2024-01-10:
Not showing on my Galaxy M20 on Android 10, Pixel 4XL on Android 13

It does appear under packages list but does not appear on launcher.
Android auto 10.9.634644
AA Torque: 2.0.21

Coops100, 2024-01-10:
> Not showing on my Galaxy M20 on Android 10, Pixel 4XL on Android 13
> 
> It does appear under packages list but does not appear on launcher. Android auto 10.9.634644 AA Torque: 2.0.21

Try in Android Auto the hamburger top right > Developer Settings, then scroll down and check Unknown Sources. This worked for me on a Pixel 2 on 11

tuononh, 2024-01-10:
Thanks @Coops100. I've just done what you said, along with clear cache of android auto, use AA AIO Tweaker to unpatch then patch again and it appears now. Not sure which step made it work :D

addyhash, 2024-01-15:
I can't seem to get connection ... I have torque pro working and displaying live data to my phone, I have AA-torque showing up in android auto but it just says connecting to ECU. My copy of torque pro is side loaded would that be the reason?

DerGue69, 2024-01-15:
On my Galaxy S22 (not rooted) it **is still not working.**

- Uninstalled AA Torque
- Cleared Android Auto Cache
- Enabled Developer Mode in Android Auto
- Allowed Unknown Sources
- Installed AA Torque

And still noting in the Car. 
Any new ideas how to get it working again on a non-rooted phone without AA Wireless?

[image]

CrimsonLory, 2024-01-15:
Galaxy s21, no way to make it work with android 14. BTW, i cannot root my device. Any suggestions to mae it work? even downgrading AA or plugin version is ok for me. Can't downgrade android nor root the device.

agronick, 2024-01-16:
> I can't seem to get connection ... I have torque pro working and displaying live data to my phone, I have AA-torque showing up in android auto but it just says connecting to ECU. My copy of torque pro is side loaded would that be the reason?

Probably an old version of Torque.

Damien0505, 2024-01-21:
Galaxy 21 Ultra 5G with Android 14, all latest updates, I have had success with getting aaTorque back in AndroidAuto by enabling dev mode on the AAWireless dongle.  Many thanks @agronick

Annunaki, 2024-01-23:
I just found this repo, and I'm having similar issue here, but my Note 20 Ultra is on Android version 13 and Android Auto version 11.0.635054, followed the instruction clearly, but the app is not showing up in the list of apps I can use with AA.

agronick, 2024-02-02:
> I just found this repo, and I'm having similar issue here, but my Note 20 Ultra is on Android version 13 and Android Auto version 11.0.635054, followed the instruction clearly, but the app is not showing up in the list of apps I can use with AA.

It is developed on a Note 20 Ultra so installing should be pretty straightforward. I'll try to make a video. Make sure you follow all the steps.

agronick, 2024-02-02:
> I just found this repo, and I'm having similar issue here, but my Note 20 Ultra is on Android version 13 and Android Auto version 11.0.635054, followed the instruction clearly, but the app is not showing up in the list of apps I can use with AA.

I made a video that should help you: https://www.youtube.com/watch?v=DDxPrPzxZ3k&ab_channel=KyleAgronick

Annunaki, 2024-02-02:
> I made a video that should help you: https://www.youtube.com/watch?v=DDxPrPzxZ3k&ab_channel=KyleAgronick

Ahh now working fine, I think I downloaded a pre release.

Thank you very much for your time and efforts.

ascendant512, 2024-02-12:
Writing to add that I bought a CarlinKit wireless adapter for AA and it did not work.  It was a lot cheaper than the AAWireless adapter, and I guess I at least have wireless Android Auto now, but no dashboard app.  
There is no setting for developer mode in the adapter's settings page.  I tested with the PC developer tools (Android Desktop Head Unit) and I do see the AATorque app there.

ascendant512, 2024-02-12:
@AnDr3w9147 please read this: https://en.wiktionary.org/wiki/threadjacking

AnDr3w9147, 2024-02-12:
My apologies.

Airislv24, 2024-02-27:
I see that there are no options to install on Android 14, except for purchasing a specific AA wireless adapter.
My chinese AA wireless adapter from Aliexpress don't have management app and I can't switch on magical developer option.

Realme GT Neo 3, latest firmware Android 14, I tried all the recommended options - AA-Torque does not appear in Android Auto anymore ...
**Important:** 
Everything worked perfectly on Android 13, and after update to Android 14 too.
But after a while I updated to latest version of AA-Torque from application, after update everything broke...

Tw1tchy, 2024-03-06:
Has anyone been able to get Developer Mode work with AAwireless on a 3rd Gen Mazda3 with the MazdaConnect infotainment v74? I can only get Pass through mode to work which disables Developer mode.

MasterPIC, 2024-06-17:
What about cheaper wireless adapters?
I'm tempted to buy a XUDA USB adapter (to solve disconnection issues caused by the USB cable) but I don't know if developer mode is available.

DIESLTEK710, 2024-06-18:
This still not working for android 14 wired??..

Im sorry but I've been searching and couldn't find an answer... But why isn't torque supported on aa? What is the reason? Is it something with the gages? Because there's only one other obd2 app that will read out values and is in the Google play app store.. But it dosnt have gages.. It just displays the info in text..

😒 I'mma sue Google for making my driving more dangerous because I have to use my phone to see my cars gages.. And can't view on my infotainment system..

DerGue69, 2024-06-19:
I purchased this AA Adapter: https://www.amazon.de/dp/B0CYQ5PG2P?psc=1&ref=ppx_yo2ov_dt_b_product_details
Adapter is working great, but not with AA Torque.

Did not find setting on the adapter to set it in DEV mode.
Any experience?

MasterPIC, 2024-06-23:
@DerGue69 

did you test the adapter only with Android 14?

DerGue69, 2024-06-24:
> @DerGue69
> 
> did you test the adapter only with Android 14?
Yes sir. Only have Android 14 phones.

MasterPIC, 2024-07-08:
I just tested a cheaper wireless adapter (by XUDA) on Android 13 and it works flawlessly. Anyway I see no developer options, so probably it won't work on Android 14. Unfortunately I've no Android 14 phone for testing.

YuhiAida, 2026-10-31:
Handled in the maintained fork: https://github.com/YuhiAida/aa-torque

The "not visible in Android Auto" symptom is an install-record problem, not app code. Android Auto admits apps from the install record (`initiatingPackageName` / `installerPackageName`), not the "Unknown sources" toggle. The fork documents the fix — reinstall through KingInstaller with Shizuku — in `docs/aa-install-gate.md`:
https://github.com/YuhiAida/aa-torque/blob/master/docs/aa-install-gate.md

Releases from the fork are signed with a stable key, so updates install in place.

xuannguyen2013, 2026-11-05:
https://www.tiktok.com/@kid_mixer/video/7703591136882904340