agronick/aa-torque #88: [BUG] [REQUEST?] Small Screen Layout Fix of 10.0.94 Scales Improperly (broke midsize screen layout)

Mtn-Mike, 2023-12-14:
**Describe the bug**
Version 10.0.94, "fix layout on smaller screens", doesn't scale properly.  See image.
[image]

I think the display from v10.0.93 is near perfect.  Maybe provide an option in the app for small, medium, and large displays that all scale a little differently, and let the user choose which fits best.

closed 2023-12-14 by agronick

Mtn-Mike, 2023-12-15:
Hey, it's back (v10.0.103)!  Nice work, thanks!

Although I just found out my S21 will be force-updated to Android 14 soon, which seems to be a death sentence for AA-Torque.  :(

anderml1955, 2023-12-15:
S21 Ultra Pro, Android 14 fully updated, latest AA & Torque. Working perfectly. Great work.

commit ac1486c6 (2023-12-14): feat: make min/max below gauge setting for large screens (fixes #88)