agronick/aa-torque #96: [BUG] "Highest value" of the dials gets reset upon loosing display focus

Bhaskar-S-Jois, 2023-12-26:
I use around 3 dashboards with various parameters and each time I navigate between dashboards, the highest value of the main 3 analog dials gets reset to whatever value is shown at the time of re-focus. I didn't disconnect any time during this, just switching dashboards is causing those values to reset. Maybe those values are not stored during the dashboard switch? Also may need to think if the values of the other non-focus dashboards will be polled before giving a fix here because let's say we have RPM parameters in 2 dashboards and each may show different highest values just because they are not polled during non-focus. Or other parameters like temperature which are critical and end up not showing the correct highest values just because they were not in focus.

agronick, 2023-12-27:
It is not stored and not polled when not focused.

agronick, 2023-12-27:
I added some code to cache it. I'm going to test it out today.

Bhaskar-S-Jois, 2023-12-27:
Thats good to know, thank you!
Although it would be good if the data is polled even when not in focus. Cause we can use a dashboard for all critical data like temperature, battery voltage etc. and peek to that dashboard occasionally to check the highest values. Since torque takes care of polling from ECU, it shouldn't be an overhead to poll the data from torque when not in focus I presume?

agronick, 2023-12-27:
It definitely is slower to request them all at once. It is staggered now. I don't know if querying every item separately would be just as fast.

closed 2023-12-28 by agronick