agronick/aa-torque #97: [BUG]

Taza13, 2023-12-28:
With 2.0.8 when I preview my guages in the app no guages show or any info
This is on Samsung S23 Ultra Android 14 

Here are a few screenshots as images
[image]
[image]
[image]

agronick, 2023-12-28:
Turn the opacity up. It will be fixed in the next version.

closed 2023-12-28 by agronick

Taza13, 2023-12-29:
Legend thanks heaps