agronick/aa-torque #102: [REQUEST]Letter for gear box instead of number ?

Vince56190, 2024-01-04:
I use AATorque to display the ratio of my gearbox but the gauges do not allow me to display letters like P R and N, is it possible to add that?

agronick, 2024-01-05:
You need a custom expression under unit conversion. This is the one I use `IF(a == 70, "P", IF(a == 60, "N", a))`. You just chain a bunch of IF else statements together, basically. See the EvalEx documentation for IF.

Vince56190, 2024-01-06:
the conversion works on the small displays but not on the main gauges, see the photos, they are the same PID with the same conversion.small one indicates P, R and N. The big one  0 ( for P and R) or 5 (for N), I suppose that the main gauges do not accept letters.
[image]

agronick, 2024-01-06:
No, they don't.

Vince56190, 2024-01-07:
Could this be added? this option could make it possible to create visual alerts for example: IF (a > 150, "ALERT", a) (I do not know if this calculation is possible with EvalEx), the gauge would indicate ALERT instead of the probe value.

agronick, 2024-01-07:
Maybe have a light or illuminate the background or something. I think it would be better to just have ranges. The EvelEx stuff is too complex for some users.

Vince56190, 2024-01-13:
Alert FAP ( DPF) done !!
[image]

agronick, 2024-01-15:
Glad you got everything working

closed 2024-01-15 by agronick