agronick/aa-torque #116: [REQUEST] Ability to relocate 'zero' position on dial

davidcharlessmith516, 2024-02-14:
**Would it be possible to have an option to relocate the 'zero' position on the dial?**

One of the PIDs in my car provides both positive and negative values, but does so asymmetrically (the negative values don't have as large of a range as the positive values). Specifically, it is the energy consumption on my electric vehicle, which goes into negative numbers when the regenerative braking kicks in. The negative values don't go as low (down to about -60 kW) as the positive values go high (up to about 265 kW). 

I can certainly set the minimum and maximum values to ensure that zero is in the center (12 o'clock) position, but was wondering if there would be any way to create an option to allow setting different minimum and maximum values while still centering the zero. A 'custom zero location' option would certainly provide the most flexibility here, but having a 'center zero' option would probably be sufficient as that would likely be the most frequent position being used.

hogass83, 2024-02-20:
You can. Just set gauge minimum to -5 or whatever you need. 0 moves accordingly.

davidcharlessmith516, 2024-02-23:
[image]

So, here's what I have right now. The left dial (currently at zero) is the one I'm referring to when discussing moving the zero position. I have different minimums and maximums for reasons explained in my first post. Between that and how the tick marks are divided up, zero isn't actually shown on the dial.

I know this is mostly an aesthetic issue, but wasn't sure if it would be possible to have the zero on the dial made visible and located at the 12 o'clock position, for instance, while maintaining the same minimum and maximum values I currently have set.

Bhaskar-S-Jois, 2024-02-24:
You can have the zero at 12 o'clock position if you give the minimums as -200 in your left hand dial case. The total divisions (in the range of minimums to maximums) of the guage is set as a perfect multiple of '8'. In your case, it is (200+80)/8 which is 35. So, a division every 35 values. But since your lower limit value starts from -80, the zero simply doesn't fall in the list as per the algorithm. 

If your main intention is to only have the zero at 12 o'clock and don't care about what the minimums is showing on the dial, just make the minimums as -200 and it will work as per your expectation.

closed 2024-04-19 by agronick