agronick/aa-torque #122: [REQUEST] Auto start

fwon2638, 2024-03-09:
Is it possible to start aa-torque automatically as soon as  Android Auto is activated?

Bhaskar-S-Jois, 2024-04-26:
Yes, this will actually help. In my car, the OBD port power shuts off if not triggered within 2 minutes or so. To activate the power back, we should shut the vehicle off fully (turning off the ECU) and power back on. For wireless AA, it requires around 60 to 90 seconds to fully boot, hence leaving a small window to activate AA-torque. If it triggers automatically, then this issue will be solved.

MartDer, 2024-10-26:
That would be great.

lm1970, 2025-02-18:
Lately the app won't update unless I open Torque. But if I stop Torque it stops AATorque. I have a S24, any solutions? Thanks.

alexbgsas, 2025-02-26:
I have the same problem with S24 Ultra for a few weeks now. AA Torque doesnt start Torque on the phone automatically. For a long time it was starting withing 10 seconds after I start the car and everything was working. When I start Torque on the phone manually the guages on the car screen start refreshing while the app is open on the phone screen. When I lock the phone the guages on the car screen keep working for 10-15 seconds and then stop refreshing.

lm1970, 2025-02-26:
For now I solved it by inserting a routine that starts Torque when the car's Bluetooth is connected and closes it when it is disconnected.

palmtreefrb, 2025-02-26:
> For now I solved it by inserting a routine that starts Torque when the car's Bluetooth is connected and closes it when it is disconnected.
> 
> Per ora ho risolto inserendo una routine che faccia partire Torque alla connessione del bluetooth dell'auto e la chiuda alla disconnessione.

How do you do this? Can you give instructions? Thanks.

lm1970, 2025-02-26:
It's simple, you just have to search for the "modes and routines" app in your s24. At least, that's the name translated from Italian, in English it might be different.

[image]