agronick/aa-torque #123: [BUG] Alarm (color) feature doesn't work with unit conversion

Bhaskar-S-Jois, 2024-03-09:
Alarm (color) feature doesn't work with unit conversion 

**Describe the bug**
The feature doesn't work when we use it under custom conversion such as time in mm:ss while getting the value in seconds. Example: FLOOR(a / 60) + ":" + FLOOR(a % 60)

In this case, the color won't change when the seconds value (which is 'a') goes above let's say 3600 (1hr). The same goes for custom made PIDs like gear indication custom PID (PRND) where P=1, R =2 and so on and used with IF condition to convert number to strings. I tried to equate the value to the number, but it didn't work.

**Media**
Screenshot attached.

**Phone**
OnePlus 11R
Tata Altroz with Harman infotainment head unit.
But I am pretty sure its a generic bug.
[image]
[image]

agronick, 2024-04-19:
Yeah, it uses the converted value which it needs to be able to parse as a number. I could have it fallback to the original value.

Bhaskar-S-Jois, 2024-04-26:
Thanks. Original value makes more sense cause we can anyway have custom PIDs from torque and also write scripts in torque app. So whatever value correction we need to make, we can make it in torque app and use it as it is in here.

Bhaskar-S-Jois, 2024-04-26:
Also one more thing I observed in the past few weeks is that the non-dial values (the 4 values which are just digital values) don't update the colors in real time. You have to switch screens or goto home and come back (basically un-focus programmatically and focus back) to get the color updated.

guirosasilva, 2024-10-01:
Good afternoon. I have a similar problem using unit conversion and PID Trip Time. The value on the display (dial) don't change, stay Zero. Using this exactly formula. How is it appears to you on the screen?

agronick, 2025-01-07:
@guirosasilva A dial will never show a string value. It needs to be a number.