agronick/aa-torque #128: [BUG] Overlap elements/graphics

Freezli, 2024-04-01:
**Describe the bug**
On the right-hand side of the head unit, icons 2 and 3 are superimposed and there are some timestamps at the bottom. On the mobile phone screen, the error is not so pronounced, icons 2 and 3 are closer together, but there are also overlaps in the timestamps.

**Media**
attached

**Phone**
Pixel 4a 5G, Android 12, latest Torque Pro and AA-Torque, USB (Aygo), AA Wireless (Yaris) connected

**Cars**
Toyota Aygo 2020 and Toyota GR Yaris 2022

[image]
[image]

agronick, 2024-04-19:
Thats strange. I'll look into it. Probably because of a smaller screen.

agronick, 2025-01-07:
Is this still happening on the latest versions? I can't reproduce it at all.