agronick/aa-torque #130: Not working on Android Auto  (2.0.26 release)

Jaafar88, 2024-04-05:
Hello there.
The newest release (2.0.26) not showing on Android Auto, any fix yet please?
Kind regards.

Vince56190, 2024-04-05:
ne fonctionne pas  avec android 14, vous devez utilisez un adapteur AAWireless

Fabio1968, 2024-08-17:
Hi everyone, I have a Samsung Flip 6 with Android 14 and I confirm that the app is not detected by Android Auto. Are there solutions foreseen? Thank you

tomblenz, 2024-09-28:
Same, Torque Pro works fine on phone but it does not appear in the Android Auto list of apps, either on the head unit itself or in the 'customise launcher' settings on phone. I went through all setup steps and the forced update appeared to complete successfully.

Google Pixel 6, Android 14. 

[image]
[image]
[image]
[image]

agronick, 2025-01-06:
Its a hack. There are ways around it. Look at the discussions page. Google won't allow it in the app store because it bypasses security restrictions.

ghost1face, 2025-01-11:
@agronick I'm not sure what security restrictions are in place.  I do develop in several languages so maybe I can assist? I'm not sure what the issue really is though.  Ping me and lets chat.  I want to get this working on my head unit in my Evo X

YuhiAida, 2026-10-31:
Handled in the maintained fork: https://github.com/YuhiAida/aa-torque

The "not visible in Android Auto" symptom is an install-record problem, not app code. Android Auto admits apps from the install record (`initiatingPackageName` / `installerPackageName`), not the "Unknown sources" toggle. The fork documents the fix — reinstall through KingInstaller with Shizuku — in `docs/aa-install-gate.md`:
https://github.com/YuhiAida/aa-torque/blob/master/docs/aa-install-gate.md

Releases from the fork are signed with a stable key, so updates install in place.