agronick/aa-torque #137: Custom Ford PID for Transmission Oil Temperature Not Updting

bmkantor, 2024-05-15:
AA-Torque display  of custom PID 221e1C [FORD] Transmission Temp does not update on the AA-Torque Display, even though it continuously updates on Torque Pro on the phone.

Phone: Pixel 7 Pro running Android 14
Car: 2019 Ford Ranger with AA Wireless version 3.50
AA-Torque version 2.0.26
Torque Pro version 1.12.101

Qu1dAg1s, 2024-05-15:
> AA-Torque display of custom PID 221e1C [FORD] Transmission Temp does not update on the AA-Torque Display, even though it continuously updates on Torque Pro on the phone.
>
> Phone: Pixel 7 Pro running [...]

Same here...

Tablet: Galaxy Tab A with S Pen Android 11, One UI Version 3.1
Car: Subaru Outback 3.6R 2019 Limited with AA version 11.8.641624-relaease
AA-Torque version 2.0.26
Torque Pro version 1.12.101

agronick, 2025-01-07:
Can you paste the logs?