agronick/aa-torque #152: [BUG] Imported DBC file in Torque not being represented in the PID list

SebastianS92, 2024-09-04:
**Describe the bug**
When importing a DBC file the sensors are being properly displayed in the torque app however they only show up as 1 sensor multiple times in aa-torque

**Media**
[image]
[image]
[Mazda Test.zip](https://github.com/user-attachments/files/16874159/Mazda.Test.zip)

**Replication**
To confirm the behaviour, download the attached zip file and import either one to your custom PID in Torque. One DBC contains the Sensors in 2 differently named can messages although sharing the same CAN-ID and the other DBC contains the sensors in the same Message. This also happens, when I put the sensors in 2 differently named messages that don't share the same CAN ID

SebastianS92, 2024-09-11:
So far I figured out it's most likely a formatting issue of the dbc file. When creating a different file for each signal it shows properly I think I've met similar issues when exporting, editing and reimporting an existing pre-configured set and my wild guess is that the culprit is the " Symbol

agronick, 2025-01-06:
Do they both have the same PID? That is probably the issue. I think we ask for names based off the list of PIDs so it is probably pulling the first one twice.