agronick/aa-torque #172: [BUG] Torque Pro doesn't launch automatically

pagrey, 2025-02-16:
The new version of Torque Pro doesn't seem to auto launch when aa-torque is used on Android 15. Version 1.12.101 does work and version 1.12.108+ doesn't.

It's easy to reproduce without connecting to Android Auto. If you configure Torque and exit the app then open aa-torque and select preview gauges the status bar will immediately show torque attempting to connect in version 1.12.101. If you repeat those steps with an identical configuration in Torque on the current version Torque will not show in the status bar and no information will be loaded. Everything does work if you launch the Torque app manually but aa-torque is not able to initiate the connection. This means that Android Auto doesn't seem to be able to connect to the adapter unless you open Torque on the phone.

I know, you have no control of what Torque does and how they change their plugin interface. I also hope I've just made a simple configuration mistake. It is unfortunate that Google and now the developer of Torque is making this so difficult.

Aptahar, 2025-02-17:
Waiting fix from author of Torque Pro.
Problem issued on official forum https://torque-bhp.com/forums/?wpforumaction=viewtopic&t=20110.0

closed 2025-02-17 by pagrey

Bhaskar-S-Jois, 2025-03-30:
Looks like Torque pro 1.12.108+ has broken a lot of things. Not sure what the torque pro devs are doing or are upto. Hope they fix whatever is broken soon so that aatorque can be used again!