agronick/aa-torque #183: [BUG] crashing when opened in android auto

iwoAntoniewicz, 2025-03-27:
| :exclamation:  Please write your issue in English  |
|-----------------------------------------|

**Describe the bug**
App crashes when opening in android auto. When opening in phone, it works fine, values are present in preview mode

**Log output**
Attached

[aaTorqueLog.txt](https://github.com/user-attachments/files/19484263/aaTorqueLog.txt)

**Phone**
Samsung s10 non 5g version
Generic android auto for bmw e60 CCC, with android 13

BloodMage87, 2025-03-27:
Mine does the same thing
I have Galaxy S9+ Android 10 and i have AAwireless with developer mode enabled
I have enabled unknown sources from developer options on android auto

RuiVil, 2025-04-03:
Exactly the same problem. Samsung Galaxy Note 9, Android 10.