agronick/aa-torque #189: [BUG] Doesn't show in Android Auto apps list on Android 15 Pixel 7.Pro

RedCloudSolutions, 2025-04-16:
All developed mode settings and install from unknown sources enabled for both phone OS and Android Auto settings. Everything seemed to install fine, but I cannot select the app from the Android Auto available apps and therefore cannot be viewed from my car.

haydoyle, 2025-04-29:
I am having the exact same issue. Worked perfectly with my Samsung Note20 Ultra

TylerHicks21, 2025-06-09:
Same issue. Android 15, Pixel 8.
Is this a Pixel issue?
Any work arounds?

bornaradusin, 2025-06-19:
You can install it using AAAD but it's a really old version, it does work up to and including android 16 :)
I've a pixel 9 xl on android 16 with the latest update

bornaradusin, 2025-06-19:
Update:

Got it working with:
https://github.com/Rikj000/Android-Auto-XLauncher-Unlocked/releases/tag/v2.0.1

1. Install it using kingstaller
2. Enable app launcher and reboot the phone
3. Connect to the car and in android auto on the car screen select all vehicle apps and then aatorque (this will add it to the AA launcher)
4. Done!!! 
Hope if helps

YuhiAida, 2026-10-31:
Handled in the maintained fork: https://github.com/YuhiAida/aa-torque

The "not visible in Android Auto" symptom is an install-record problem, not app code. Android Auto admits apps from the install record (`initiatingPackageName` / `installerPackageName`), not the "Unknown sources" toggle. The fork documents the fix — reinstall through KingInstaller with Shizuku — in `docs/aa-install-gate.md`:
https://github.com/YuhiAida/aa-torque/blob/master/docs/aa-install-gate.md

Releases from the fork are signed with a stable key, so updates install in place.