agronick/aa-torque #195: [REQUEST] [BUG] v2.0.30 Doesn't show in Android Auto apps list on Android 15 multiple phone brands

wrakdzdr, 2025-05-20:
Hi first, thank you! 

I installed all without a problem but v2.0.30 doesn't show in Android Auto apps list on Android 15. It works on the mobile device review but can't get it up on the car screen since it doesn't show in the AA apps list, therefore no icon on the screen

Thank you. I will sponsor the project once I get it working. Cheers!

samuelleung83, 2025-06-09:
Confirmed I am experiencing this issue as well.

TylerHicks21, 2025-06-09:
Same issue. Any work around?

bornaradusin, 2025-06-19:
You can install it using AAAD but it's a really old version, it does work up to and including android 16 :)

Allonsbon2004, 2025-06-24:
I try it without succes. I have not option All vehicule apps on my car tab

bornaradusin, 2025-06-24:
> No I have not LSPOSED

You'll need that to get this working, i use this one as it works nice with gphotos and other mods i use
https://github.com/JingMatrix/LSPosed

bornaradusin, 2025-06-24:
> It can RUN on no ROOT phone ?

Not that i know of unfortunately.

SimonLyke, 2025-07-12:
> It can RUN on no ROOT phone ?

i was running it on a non rooted S24+, however it only worked for a short period of time as app no longer shows up in the list of available apps for android auto

lstefunek, 2025-07-19:
aatorque doesn't work on AAAD app yet - waiting for a fix

Allonsbon2004, 2025-07-21:
Hi,
Go here. Tested Ok on Android 15.
https://github.com/Inside4ndroid/aa-torque-ex/discussions/2

YuhiAida, 2026-10-31:
Handled in the maintained fork: https://github.com/YuhiAida/aa-torque

The "not visible in Android Auto" symptom is an install-record problem, not app code. Android Auto admits apps from the install record (`initiatingPackageName` / `installerPackageName`), not the "Unknown sources" toggle. The fork documents the fix — reinstall through KingInstaller with Shizuku — in `docs/aa-install-gate.md`:
https://github.com/YuhiAida/aa-torque/blob/master/docs/aa-install-gate.md

Releases from the fork are signed with a stable key, so updates install in place.

SimonLyke, 2026-11-01:
> requests on the go with GitHub Mobile for iOS
> and Android
> Download it today!

To be honest. I'm not touching this because you vibe coded it

YuhiAida, 2026-11-01:
> 16\. To be honest. I'm not touching this because you vibe coded it
>      […](#)
>       <[#195 (comment)](https://github.com/agronick/aa-torque/issues/195#issuecomment-6548225180)>
>      
>       Handled in the maintained fork: https://github.com/YuhiAida/aa-torque
>      
>       The "not visible in Android Auto" symptom is an install-record problem,
>       not app code. Android Auto admits apps from the install record (
>       initiatingPackageName / installerPackageName), not the "Unknown sources"
>       toggle. The fork documents the fix — reinstall through KingInstaller with
>       Shizuku — in docs/aa-install-gate.md:
>       https://github.com/YuhiAida/aa-torque/blob/master/docs/aa-install-gate.md
>      
>       Releases from the fork are signed with a stable key, so updates install in
>       place.
>      
>       <[#195](https://github.com/agronick/aa-torque/issues/195)?email_source=notifications&email_token=AWQDVE2GT3W3DEZACIQIOWD5WY55NA5CNFSNUABFM5UWIORPF5TWS5BNNB2WEL2JONZXKZKDN5WW2ZLOOQXTMNJUHAZDENJRHAYKM4TFMFZW63VHMNXW23LFNZ2KKZLWMVXHJLDGN5XXIZLSL5RWY2LDNM#issuecomment-6548225180>,
>       requests on the go with GitHub Mobile for iOS
>       and Android
>       Download it today!

https://github.com/YuhiAida/aa-torque/commit/fc78005ed9d155e97135806ccbed3ae9eeeaeeb9