agronick/aa-torque #205: Android 15 and samsung

XTho-masX, 2025-09-24:
Hi,
I have a Samsung S23 with Android Auto 15, but I don't see AA-Torque among the available apps. I followed the procedure as per the videos I found and enabled unknown sources on AA, but nothing happened. Can anyone help me? Thanks.

hiagocosta, 2025-10-08:
me too, cant use aa-torque

todobarato, 2025-12-14:
I have android 15, same issue, no available app icon for AA-Torque, maybe instructions for rooted phone? 
Thanks.

kwc0810, 2026-04-10:
@SyCoREAPER Hi, I'm trying to get a CATEGORY_PROJECTION app working on the latest Android Auto and struggling. Would appreciate your help.

My setup:
- Device: Samsung SM-X135F (tablet, Android 16)
- Android Auto: 16.5.661214-release
- Google Play Services: 26.12.33
- DHU 2.0 (2022-03-30 build)

What I tried:

1. ADB PackageInstaller Session API with Play Store installer spoofing
   pm install-create -i com.android.vending --install-reason 1 -g → install-write → install-commit
   Confirmed installer=com.android.vending via pm list packages -i
   → Result: CATEGORY_PROJECTION app not visible in AA (both DHU and real car head unit)

2. Android Auto downgrade (7.8 and 12.0)
   Factory reset via cmd package uninstall -k, then installed older AA APK
   → Result: certificate has expired / security check error 14 (incompatible with latest Play Services)

3. Shizuku + InstallWithOptions
   → Shizuku won't install on my device: "This app was built for an older version of Android"

Questions:
- When you got it working on S25 Ultra + Android 16, what Android Auto version were you running? Did it work on 16.5?
- Does InstallWithOptions via Shizuku do anything differently than ADB's pm install-create -i com.android.vending --install-reason 1? Any additional flags?
- Is CATEGORY_PROJECTION only visible on real car head units and never on DHU?

Any guidance would be greatly appreciated. Thanks!

mitlaendre, 2026-10-20:
I was finally able to install it so the app is visible in AA. I used AAEnabler and Shizuka.

YuhiAida, 2026-10-31:
Handled in the maintained fork: https://github.com/YuhiAida/aa-torque

The "not visible in Android Auto" symptom is an install-record problem, not app code. Android Auto admits apps from the install record (`initiatingPackageName` / `installerPackageName`), not the "Unknown sources" toggle. The fork documents the fix — reinstall through KingInstaller with Shizuku — in `docs/aa-install-gate.md`:
https://github.com/YuhiAida/aa-torque/blob/master/docs/aa-install-gate.md

Releases from the fork are signed with a stable key, so updates install in place.